DavidEGrayson/minimu9-ahrs #16: minimu9-ahrs.1/COORDINATE SYSTEMS: missing note about circular direction

nxdefiant, 2015-02-16:
I noticed that the first angle is inverted when using "gyro-only". Is that on purpose? Seems natural since clockwise is the natural direction of the compass while counter clock is L3G. Still confusing... Could you maybe drop a note in the "COORDINATE SYSTEMS"?

DavidEGrayson, 2015-02-16:
I am not sure what the problem is.  It sounds like you are using the `--output euler` and `--gyro-only` options, and the first angle (pitch) not is inverted from what you would expect.

The "ORIENTATION AS EULER ANGLES" section of the man page explains what all the angles are, by walking you through a transformation from the ground coordinate system to the body (circuit board) coordinate system.  The second rotation uses the pitch angle:

> Now rotate the coordinate system about its second axis by a certain
> angle that we will call the *pitch*.  A positive pitch makes
> the first axis of the coordinate system go up.  A pitch of 90.0
> would make the first axis point straight up, while a pitch of -90.0
> would make the first axis point straight down.

The ideas of positive pitch and negative pitch don't really map that well to the words "clockwise" and "counter clockwise", so I am not sure what the issue is.

nxdefiant, 2015-02-17:
It is simple that I would like to see the rotation direction documented as e.g. clockwise/counterclockwise for the different options.

DavidEGrayson, 2015-02-17:
Sorry, I don't get it.  The first angle is pitch.  On a plane, pitch describes how far your plane is from pointing at the horizon.  Are you sure you are trying to talk about pitch?  Maybe you actually want to talk about yaw, which is the second angle.  Or perhaps the man page is wrong, and yaw is actually the first angle?

nxdefiant, 2015-02-18:
Right, there seems to be a problem, too. I'm using minimu9-ahrs on an AltIMU-10 v3, the first value is yaw for me, as described under "--output FORMAT"