DavidEGrayson/minimu9-ahrs #18: timing with polling vs timerfd

nxdefiant, 2015-02-16:
In minimu9-ahrs.cpp I noticed that you are polling for your timing:
```c
        // Ensure that each iteration of the loop takes at least 20 ms.
        while(millis() - start < 20) 
        {   
            usleep(1000);
        }  
```
Please consider using timerfd() instead (see man 2 timerfd_create) since it will just wake the thread at the right time (well more or less since we do not have a rtos...). If requested I can do a pull request.

Also using the Fifo-scheduler should help with the timing, eg. pthread_setschedparam(pthread_self(), SCHED_FIFO, ...)

DavidEGrayson, 2015-02-16:
Thanks, that sounds interesting.  I will look into using timerfd the next time I am working on this project.

DavidEGrayson, 2017-04-16:
Thanks for the tip about the timerfd.  I implemented it today on a branch that is not merged into master yet and it seems to improve the timing a lot.  The time between loop iteration starts, as measured by `std::chrono::steady_clock`, went from an average of **20051702 ns (0.25% slow)** with a standard deviation of **386895 ns (1.9%)** to an average of **20000006 ns** with a standard deviation of *704 ns*.  These tests were on a Raspberry Pi 3 that was not busy with other tasks.

As for your second suggestion, it looks like using `SCHED_FIFO` or `SCHED_RR` requires root privileges on Raspbian.  I tried both the POSIX call you suggested, and the corresponding Linux system call.  This is the code I tried to use to do it:

  struct sched_param param;
  param.sched_priority = 10;
  int sched_result = sched_setscheduler(0, SCHED_FIFO, &param);
  if (sched_result == -1)
  {
    throw posix_error("Could not set priority");
  }

closed 2017-04-16 by DavidEGrayson

nxdefiant, 2017-04-17:
This can be configure in /etc/security/limits.conf, a line like

`@realtime	-	rtprio		99`

would give everyone in the realtime group the privilege for the rt-schedulers up to prio 99.

commit 80149ecd (2017-04-16): Wrap up the timerfd code in a nice new class called pacer.

Fixes #18.