DavidEGrayson/minimu9-ahrs #19: Adding support for minimu-9 v5

nathanielatom, 2016-03-09:
What kind of changes would be necessary for adding support for [v5](https://www.pololu.com/product/2738)? It uses [LSM6DS33](https://www.pololu.com/file/download/LSM6DS33.pdf?file_id=0J1087) and [LIS3MDL](https://www.pololu.com/file/download/LIS3MDL.pdf?file_id=0J1089). Would it just be a matter of making changes analogous to this [commit](https://github.com/DavidEGrayson/minimu9-ahrs/commit/ec9a75539d89e1166ed7fcb28acdd611c0b8a103)? Or would something else be involved? 

Thanks!

DavidEGrayson, 2016-03-12:
Hi.  Thanks for nudge, I should probably update my code some time.  I don't have a time frame for when I might do this, so that's cool if you give it a shot.  Updating to the MinIMU-9 v5 might be tricky, because you would have to add support for two new *classes* of chips instead of just adding support for two new *models*.  You would probably need to add a LSM6 class and a LIS3MDL class or something.  Getting everything to be auto-detected might be tricky.  But you can refer to our official Arduino libraries as a reference:

https://github.com/pololu/lsm6-arduino
https://github.com/pololu/lis3mdl-arduino

Aaron-Davison, 2016-07-10:
Hi. Has any progress been made on this? Or is there another option to getting v5 to work on Pi 3?
Cheers

nathanielatom, 2016-07-10:
Hi, since development time was limited for our project, we swapped to v3 and used https://github.com/RPi-Distro/RTIMULib/blob/master/README.md, with good results.

DavidEGrayson, 2016-07-10:
No progress has been made on MinIMU-9 v5 support as far as I know, but you can still buy the [v3](https://www.pololu.com/product/2468) and use it with the code in this repository.

DavidEGrayson, 2017-04-16:
The version on master now supports the MinIMU-9 v5, and I will soon tag it as version 3.0.0.

closed 2017-04-16 by DavidEGrayson