DavidEGrayson/minimu9-ahrs #20: Make Error: undefined reference to boost::program_options

ahmedfarazkhan, 2016-03-13:
When I tried to `make` the project, I encountered two errors with destructor scope in I2CBus.cpp and MinIMU9.cpp that were resolved by including `<unistd.h>`. However, I now get a new error due to undefined references to functions from _boost::program_options_. I suspect this is a linking issue (I have installed boost on my system), but the makefile contains the line `LDFLAGS += -lboost_program_options`, so I'm not sure how to proceed. A snippet of the `make` output is included below:

```
minimu9-ahrs.o: In function `boost::program_options::variables_map::~variables_map()':
minimu9-ahrs.cpp:(.text._ZN5boost15program_options13variables_mapD0Ev[_ZN5boost15program_options13variables_mapD0Ev]+0xe): undefined reference to `vtable for boost::program_options::variables_map'
minimu9-ahrs.o:(.rodata._ZTIN5boost15program_options20multiple_occurrencesE[_ZTIN5boost15program_options20multiple_occurrencesE]+0x10): undefined reference to `typeinfo for boost::program_options::error_with_option_name'
minimu9-ahrs.o:(.rodata._ZTIN5boost15program_options11typed_valueISscEE[_ZTIN5boost15program_options11typed_valueISscEE]+0x18): undefined reference to `typeinfo for boost::program_options::value_semantic_codecvt_helper<char>'
minimu9-ahrs.o:(.rodata._ZTVN5boost15program_options11typed_valueISscEE[_ZTVN5boost15program_options11typed_valueISscEE]+0x38): undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::string, std::allocator<std::string> > const&, bool) const'
collect2: error: ld returned 1 exit status
make: *** [minimu9-ahrs] Error 1

```

DavidEGrayson, 2016-03-13:
Sorry, I'm not sure what's going wrong, and I haven't tried compiling this code in a while.  Could you show a little bit more output from Make so we can see the gcc command that is failing?  In particular, I want to make sure that `-lboost_program_options` comes after the other things that are being linked in.

ahmedfarazkhan, 2016-03-14:
This is the entire output:
```
g++ -lboost_program_options  minimu9-ahrs.o I2CBus.o L3G.o LSM303.o MinIMU9.o   -o minimu9-ahrs
minimu9-ahrs.o: In function `boost::program_options::typed_value<std::string, char>::xparse(boost::any&, std::vector<std::string, std::allocator<std::string> > const&) const':
minimu9-ahrs.cpp:(.text._ZNK5boost15program_options11typed_valueISscE6xparseERNS_3anyERKSt6vectorISsSaISsEE[_ZNK5boost15program_options11typed_valueISscE6xparseERNS_3anyERKSt6vectorISsSaISsEE]+0x19): undefined reference to `boost::program_options::validate(boost::any&, std::vector<std::string, std::allocator<std::string> > const&, std::string*, int)'
minimu9-ahrs.o: In function `boost::program_options::typed_value<std::string, char>::name() const':
minimu9-ahrs.cpp:(.text._ZNK5boost15program_options11typed_valueISscE4nameEv[_ZNK5boost15program_options11typed_valueISscE4nameEv]+0xd): undefined reference to `boost::program_options::arg'
minimu9-ahrs.o: In function `boost::program_options::basic_command_line_parser<char>::basic_command_line_parser(int, char const* const*)':
minimu9-ahrs.cpp:(.text._ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc[_ZN5boost15program_options25basic_command_line_parserIcEC5EiPKPKc]+0x1e0): undefined reference to `boost::program_options::to_internal(std::string const&)'
minimu9-ahrs.cpp:(.text._ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc[_ZN5boost15program_options25basic_command_line_parserIcEC5EiPKPKc]+0x231): undefined reference to `boost::program_options::detail::cmdline::cmdline(std::vector<std::string, std::allocator<std::string> > const&)'
minimu9-ahrs.cpp:(.text._ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc[_ZN5boost15program_options25basic_command_line_parserIcEC5EiPKPKc]+0x473): undefined reference to `boost::program_options::to_internal(std::string const&)'
minimu9-ahrs.o: In function `boost::program_options::basic_command_line_parser<char>::run()':
minimu9-ahrs.cpp:(.text._ZN5boost15program_options25basic_command_line_parserIcE3runEv[_ZN5boost15program_options25basic_command_line_parserIcE3runEv]+0x1d): undefined reference to `boost::program_options::detail::cmdline::get_canonical_option_prefix()'
minimu9-ahrs.cpp:(.text._ZN5boost15program_options25basic_command_line_parserIcE3runEv[_ZN5boost15program_options25basic_command_line_parserIcE3runEv]+0x55): undefined reference to `boost::program_options::detail::cmdline::run()'
minimu9-ahrs.o: In function `boost::program_options::variables_map::~variables_map()':
minimu9-ahrs.cpp:(.text._ZN5boost15program_options13variables_mapD2Ev[_ZN5boost15program_options13variables_mapD5Ev]+0xe): undefined reference to `vtable for boost::program_options::variables_map'
minimu9-ahrs.o: In function `main':
minimu9-ahrs.cpp:(.text.startup+0x1b): undefined reference to `boost::program_options::options_description::m_default_line_length'
minimu9-ahrs.cpp:(.text.startup+0x72): undefined reference to `boost::program_options::options_description::options_description(std::string const&, unsigned int, unsigned int)'
minimu9-ahrs.cpp:(.text.startup+0x147): undefined reference to `boost::program_options::options_description::add_options()'
minimu9-ahrs.cpp:(.text.startup+0x161): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, char const*)'
minimu9-ahrs.cpp:(.text.startup+0x173): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, char const*)'
minimu9-ahrs.cpp:(.text.startup+0x188): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
minimu9-ahrs.cpp:(.text.startup+0x19d): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
minimu9-ahrs.cpp:(.text.startup+0x1b2): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
minimu9-ahrs.cpp:(.text.startup+0x1f0): undefined reference to `boost::program_options::variables_map::variables_map()'
minimu9-ahrs.cpp:(.text.startup+0x214): undefined reference to `boost::program_options::detail::cmdline::set_options_description(boost::program_options::options_description const&)'
minimu9-ahrs.cpp:(.text.startup+0x243): undefined reference to `boost::program_options::store(boost::program_options::basic_parsed_options<char> const&, boost::program_options::variables_map&, bool)'
minimu9-ahrs.cpp:(.text.startup+0x265): undefined reference to `boost::program_options::notify(boost::program_options::variables_map&)'
minimu9-ahrs.cpp:(.text.startup+0x2c3): undefined reference to `boost::program_options::operator<<(std::ostream&, boost::program_options::options_description const&)'
minimu9-ahrs.cpp:(.text.startup+0x7d4): undefined reference to `boost::program_options::error_with_option_name::get_canonical_option_name() const'
minimu9-ahrs.o: In function `boost::program_options::variables_map::~variables_map()':
minimu9-ahrs.cpp:(.text._ZN5boost15program_options13variables_mapD0Ev[_ZN5boost15program_options13variables_mapD0Ev]+0xe): undefined reference to `vtable for boost::program_options::variables_map'
[... cut ...]
minimu9-ahrs.o:(.rodata._ZTIN5boost15program_options11typed_valueISscEE[_ZTIN5boost15program_options11typed_valueISscEE]+0x18): undefined reference to `typeinfo for boost::program_options::value_semantic_codecvt_helper<char>'
minimu9-ahrs.o:(.rodata._ZTVN5boost15program_options11typed_valueISscEE[_ZTVN5boost15program_options11typed_valueISscEE]+0x38): undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::string, std::allocator<std::string> > const&, bool) const'
collect2: error: ld returned 1 exit status
make: *** [minimu9-ahrs] Error 1

```

DavidEGrayson, 2016-04-26:
Hey, sorry for forgetting about this.

Could you try running this in your terminal after you get the error message above?

~~~~
g++ minimu9-ahrs.o I2CBus.o L3G.o LSM303.o MinIMU9.o -lboost_program_options -o minimu9-ahrs
~~~~

Moving the library to the end will probably allow G++ to find those symbols.  I'm not sure why the code compiled fine for me in 2012 though, and I'm not sure why you are the only one who has reported this.

closed 2017-04-16 by DavidEGrayson

vishalkumarchaudhary, 2018-09-30:
Can we make this on cpu not gpu ?

DavidEGrayson, 2018-09-30:
minimu9-ahrs does not use the GPU.  You might be thinking of it's sister project, which is called ahrs-visualizer, and uses the GPU.  Either way, your post is off topic for this thread since it has nothing to do with the original issue at the top.

commit 5ada2023 (2017-04-16): Fix the way boost_program_options is linked.

Fixes #20.