DavidEGrayson/minimu9-ahrs #21: minimu-ahrs-calibrate fails

rsend, 2016-04-26:
i tried to calibrate the magnetometer so i could use the visualizer (raw mode works great), but I'm getting the following Python error:

Optimizing calibration...Traceback (most recent call last):
  File "/usr/bin/minimu9-ahrs-calibrator", line 147, in <module>
    run()
  File "/usr/bin/minimu9-ahrs-calibrator", line 101, in run
    full_output=False, disp=verbose, retall=False)
  File "/usr/lib/python2.7/dist-packages/scipy/optimize/optimize.py", line 373, in fmin
    res = _minimize_neldermead(func, x0, args, callback=callback, **opts)
  File "/usr/lib/python2.7/dist-packages/scipy/optimize/optimize.py", line 438, in _minimize_neldermead
    fsim[0] = func(x0)
  File "/usr/lib/python2.7/dist-packages/scipy/optimize/optimize.py", line 281, in function_wrapper
    return function(*(wrapper_args + args))
TypeError: can only concatenate tuple (not "list") to tuple

This is running ona Pi Zero, if that helps...

DavidEGrayson, 2016-04-26:
Some people were [discussing this on my blog](http://blog.davidegrayson.com/2012/11/orientation-sensing-with-raspberry-pi.html?showComment=1439668636650#c5173827649123192014).  The code worked the last time I tried it, but apparently something changed and you need to edit line 100 of `/usr/bin/minimu9-ahrs-calibrator`, changing

    args=[raw_readings], 

to

    args=(raw_readings,)

I don't really like that script though, I need to change it to be more robust.

rsend, 2016-04-27:
That seems to work for the moment. Note that I needed to ADD a comma on the inside, not just move it; like so: args=(raw_readings,),  ....

tomit89, 2016-06-29:
Hi,
I'am having the exact same issue, I've run:
sudo minimu9-ahrs-calibrate -b /dev/i2c-1

and I'am getting exact same error "TypeError: can only concatenate tuple (not "list") to tuple"
I've modified 100th line of minimu9-ahrs-calibrator and tried all combinations (with two commas and one), Unfortunatly problem remains.

I'm using raspberry pi 2 B. Os info:
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

Thank you for your reply.

DavidEGrayson, 2016-06-29:
@tomit89 Can you paste the full output of `sudo minimu9-ahrs-calibrate -b /dev/i2c-1` and also post the full output of `grep -n raw_readings /usr/bin/minimu9-ahrs-calibrator`?

tomit89, 2016-06-30:
When I've ran grep -n raw_readings /usr/bin/minimu9-ahrs-calibrator, I found out that I had another calibrator file somewhere. When I fixed the 100th line there it started working.

Thanks.

thuna-garry, 2016-10-02:
Also had the same issue on pi-zero.
Hand edit as suggested fixed the problem.

animeshpanara, 2017-01-01:
i changed the 100 th line error got solved but it doesnt get calibrated it always shows optimizing calibration then shows wrong data set optimising failed try again with new pair of data set

DavidEGrayson, 2017-01-01:
Hey, you are experiencing issue #14 I think.  I have not gotten around to fixing this yet, sorry.  For now you might want to just look at the raw numbers and manually record the minimum and maximum readings for the X, Y, and Z axes in order to generate your `.minimu9-ahrs-cal` calibration file.  The format of the file is described in the minimu9-ahrs man page, type `man minimu9-ahrs` or view it [here](https://github.com/DavidEGrayson/minimu9-ahrs/blob/master/minimu9-ahrs.1).

DavidEGrayson, 2017-04-16:
On the current master branch, I have replaced the calibration script with something much simpler  that should always work.

closed 2017-04-16 by DavidEGrayson