DavidEGrayson/minimu9-ahrs #23: minimu9-ahrs --mode raw

NarcisaC, 2017-03-30:
Hello, I have followed the steps of the tutorial but just when I run "minimu9-ahrs --mode raw" this output "segmentation fault (core dumped)". How do I solve this?
Also when executing "minimu9-ahrs-calibrate -b / dev / i2c-1" this other problem comes out "To calibrate the magnetometer, start rotating the IMU through
As many different orientations as possible.
Reading data ... Segmentation fault (core dumped)
 Done
Warning: Only 0 readings were provided.
Optimizing calibration ... Traceback (most recent call last):
  File "/ usr / bin / minimu9-ahrs-calibrator", line 147, in <module>
    Run ()
  File "/ usr / bin / minimu9-ahrs-calibrator", line 95, in run
    Cal1 = guess (raw_readings)
  File "/ usr / bin / minimu9-ahrs-calibrator", line 127, in guess
    Guess.extend (percentile_to_value (values, 1, 99))
  File "/ usr / bin / minimu9-ahrs-calibrator", line 18, in percentile_to_value
    Return [list [int (p / 100.0 * (len (list) -1))] for p in percentiles]
IndexError: list index out of range "

I am using:
MinIMU-9 v3 and Raspberry Pi 3.

Thanks in advance.

DavidEGrayson, 2017-03-31:
Hmm, the segmentation fault is odd.  How did you install minimu9-ahrs (e.g. did you compile it from source or use my Debian package)?

One of the other error messages you are getting is caused by issue #21, so see that issue for a workaround.

NarcisaC, 2017-03-31:
Debian package

DavidEGrayson, 2017-03-31:
Ok, could you try downloading the latest version of the code from GitHub and compiling it from source?  It is possible that something about the shared libraries ha a changed, and it will be easier to diagnose and fix bugs if you can compile it from source.

DavidEGrayson, 2017-03-31:
You would just need to run a couple command a that are listed in the README:

https://github.com/DavidEGrayson/minimu9-ahrs/blob/master/README.md

NarcisaC, 2017-04-03:
I installed putty on my laptop and ran from there but using:
minimu9-ahrs --mode raw -b / dev / i2c-1

And then it came out, then I ran it back on the raspberry and it also came out.

There is an observation: it does not always work well, it often comes out "Segmentation fault (core dumped)

DavidEGrayson, 2017-04-04:
So did you compile it from source like I suggested?  If so, we can try using gdb to see what is wrong.

closed 2017-04-15 by DavidEGrayson