DavidEGrayson/minimu9-ahrs #27: Build error on archlinux

gaetanww, 2018-02-14:
Hi,

I'm trying to build minimu9-ahrs on archlinux.
I get the following errors while building:
`i2c_bus.cpp: In member function 'void i2c_bus::write_byte_and_read(uint8_t, uint8_t, uint8_t*, size_t)':`
`i2c_bus.cpp:80:38: error: invalid use of incomplete type 'struct i2c_msg'`
`     { address, 0, 1, (typeof(i2c_msg().buf)) &command },`

`In file included from i2c_bus.cpp:8:0:`
`/usr/include/linux/i2c-dev.h:66:9: note: forward declaration of 'struct i2c_msg'`
`struct i2c_msg *msgs; /* pointers to i2c_msgs */`

Did I make a mistake while installing the headers or is that something else?

DavidEGrayson, 2018-02-14:
The Linux developers must have made some breaking changes in their header files.  Could you try adding
`#include <linux/i2c.h>` near the top of that file?

gaetanww, 2018-02-15:
I tried what you suggested and the build worked. Thank you very much ! I also had to add 
`#include <sys/ioctl.h>` on top of i2c_bus.h.

andrewssobral, 2019-05-26:
Great! thank you @DavidEGrayson @gaetanww, it worked fine for me too on my Jetson TX2.
Just added:
```
i2c-bus=/dev/i2c-1
```
in `~/.minimu9-ahrs`

DavidEGrayson, 2023-07-11:
This was fixed by c42cca6c225d47def454c41542accd36ded454da .