DavidEGrayson/minimu9-ahrs #30: Trouble detecting sensors MinIMU-9 v5

amerkay, 2018-09-18:
Hi,

I'm trying to get MinIMU-9 v5 (purchase page mentions LSM6DS33 and LIS3MDL). I keep getting "no sensor detected" even though I see it when I run `i2cdetect -y 1`.

Am I missing something? Any help would be appreciated.

Thanks for putting up the code, this really helps!

SeanHoang2203, 2018-09-28:
Hi, 

I have the same issues. this is the result from ` minimu9-ahrs --mode raw` 

Error: No magnetometer found.
Error: No accelerometer found.
Error: No gyro found.
Error: Needed sensors are missing.

This is the results from `sudo i2cdetect -y 1`
`     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- 3b 3c 3d -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                       `

I am using Tinker Board S, TinkerOS (Debian derivatives) and AltIMU 10v5

Thanks

DavidEGrayson, 2018-09-28:
You are not using the right i2c bus when you run minimu9-ahrs.  Try adding the `-b /dev/i2c-1` option.

closed 2018-09-28 by DavidEGrayson

DavidEGrayson, 2018-09-28:
What exact hardware do you have?  This thread is about the MinIMU-9 v5 but I don't see either address for its LSM6DS33 in your listing from i2cdetect.  If you are not getting the right addresses in i2cdetect, you need to troubleshoot that first.

amerkay, 2018-09-29:
My problem turned out to be that ODROID XU4 needs a Shifter Shield to get the I2C voltages converted from 3.3V -> 1.8V. 

Thanks for the offer to help anyway!