DavidEGrayson/minimu9-ahrs #33: Support ENU

jordanjohnp, 2019-02-15:
Thanks for your excellent work on 'minimu9-ahrs'. I'm creating a ROS driver from it and it's basically working, especially if I use 'fuse_gyro_only()'. But when running with 'fuse_default()', the robot appears upside down (in the ROS rviz visualizer). Also, rotations of the robot about the x-axis sees changes in the z-axis (in the visualization).

ROS uses ENU, whereas minimu9-ahrs uses NED. So I tried changing the code like this (see "// JJ"):

    /*
      doesn't work...
      we want East North Up (ENU), but existing is NED... how to convert?
    */
    matrix rotation_from_compass(const vector & acceleration, const vector & magnetic_field)
    {
      vector up = acceleration; // JJ - removed negation
      vector east = up.cross(magnetic_field); // actually it's magnetic east
      vector north = east.cross(up);

      east.normalize();
      north.normalize();
      up.normalize();

      matrix r;
      r.row(0) = east; // JJ - swapped these
      r.row(1) = north;
      r.row(2) = up;
      return r;
    }

Could you offer some advice about how I should change the code to support ENU?

Thanks for any hints!

DavidEGrayson, 2019-02-18:
Instead of changing all the code everywhere to use a new coordinate system, I think you should just do a matrix multiplication in `output_matrix` that converts it to use the new coordinate system.  You will have to study the minimu9-ahrs man page to understand what each cell of the matrix represents, and then come up with the appropriate matrix (consisting of zeros, ones, and negatives ones) that converts it to use the different coordinate system, and figure out whether to multiply that on the left or right.  I don't happen to know the answers to all those questions.

jordanjohnp, 2019-02-19:
Thanks for the comment. I'll give it a shot.