DavidEGrayson/minimu9-ahrs #43: error while building minimu9-ahrs

wawawan, 2021-05-01:
Hi,
I am trying to install minimu9-ahrs on my raspberry pi3 but I find some error with "eigen" : 

`Package eigen3 was not found in the pkg-config search path.
Perhaps you should add the directory containing `eigen3.pc'
to the PKG_CONFIG_PATH environment variable
No package 'eigen3' found
g++  -I.  -Wall --std=gnu++11 -O2 -MD -MP vector.h -o vector.h.gch
vector.h:5:10: fatal error: Eigen/Core: Aucun fichier ou dossier de ce type
 #include <Eigen/Core>
          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:37: vector.h.gch] Error 1`

could you please help me resolving this problem ? thanks

DavidEGrayson, 2021-05-02:
Did you see this part in the README?

> To build minimu9-ahrs, you first need to isntall some libraries that it depends on. On Raspbian, you can install these dependencies by running:
>
>     sudo apt-get install libi2c-dev libeigen3-dev libboost-program-options-dev

wawawan, 2021-05-02:
Yes I did it, I try again...
Yes it seems that eigen3 is now well installed !
I fellow the tutorial.
Thanks you !

closed 2021-05-02 by DavidEGrayson