DavidEGrayson/minimu9-ahrs #44: No sensors found

wawawan, 2021-05-02:
Hello, thank you for helping me, 
I need some help again.

So, I fellow some tutorials, all seems to be ok :  
- i2c is running on my raspberri pi3 ( /dev/i2c-1 )
- all libraries are installed
- sensors are detected when I run   "  i2cdetect -y 1  "   exactly as you described in your tutorial
- I build minimu9-ahrs as described, with " make " and " sudo make install "  ( no error messages )

But now when I try to run minimu9-ahrs the response is : 
- Error : No magnetometer found
- Error : No accelerometer found
- Error : No gyro found
- Error : Needed sensors are missing

Is a test to find where the mistake is ? 
Thanks a lot.

wawawan, 2021-05-02:
Is it possible these errors occured because of bad setting device permissions ?
thanks.

DavidEGrayson, 2021-05-02:
What method did you use to tell minimu9-ahrs to use `/dev/i2c-1`?  It uses `/dev/i2c-0` by defaut.

wawawan, 2021-05-03:
Hi,
first, when all was configurated I try " $ minimu9-ahrs --mode raw " and the result was " bad BUS i2c-0 " or something like that.
I wasn't surpised because I use the i2c_1 bus and because i2c-0 is the defaut Bus.
So I modified the file /usr/bin/prog_option.cpp in order to use the i2c-1 bus (In the first lines I change de Bus).
After that I try again " $ minimu9-ahrs --mode raw " and the response was :

    Error : No magnetometer found
    Error : No accelerometer found
    Error : No gyro found
    Error : Needed sensors are missing

But maybe the method was wrong...

DavidEGrayson, 2021-05-04:
That's unusual that you'd have a file named `prog_option.cpp` in `/usr/bin`, which is normally used binaries.  If I were you, I'd double check that you edited the correct file, and that you recompiled the program after doing so.  You can run `strace minimu9-ahrs --mode raw` to see what files it opens (the I2C bus is represented as a file in Linux).  You can also add `-b /dev/i2c-1` to the command when you run `minimu9-ahrs`.

wawawan, 2021-05-06:
Hi, 
So I did it again, and rebuild the minimu-ahrs without changing prog_option.cpp file.
I run minimu9-ahrs adding -b /dev/i2c-1
Results are the same as previously (or little bit different because I don't have the line " Error : No magnetometer found") : 

Error : No accelerometer found
Error : No gyro found
Error : Needed sensors are missing

[image]
[image]

DavidEGrayson, 2021-05-06:
You only showed the first page of the `strace` output, where it was busy reading a shared library.  You would have to scroll down more to look for the place where the real code starts runing, and you should be able to see an `open` command where it opens a `/dev/i2c*` file.

Instead of running `strace`, maybe a  quicker idea is to run `strings /usr/bin/minimu9-ahrs | grep /dev/i2c`.  That should show you exactly which bus is the default bus, without having to scroll down through lots of `strace` output.  If it says `/dev/i2c-0`, then you definitely failed to recompile and/or reinstall the program with your new default bus.

Let's get that working, because if you are unable to recompile the code then it's going to be very difficult for you to get things working, or try to debug things.

DavidEGrayson, 2021-05-06:
OK, here's what I think you should specifically do:

1. Delete `/usr/bin/prog_options.cpp` and any other weird files you installed in `/usr/bin/` that don't belong there.  That directory is for binaries.
2. Delete `/usr/bin/minimu9-ahrs` so we don't accidentally run it.  We need to focus on getting the program to work first, before we install it into `/usr/bin`.  
3. Go to the directory where you have the minimu9-ahrs source code.
4. Check the README.md file there, scroll to the "Version history", and make sure it has an entry for "3.0.0" there.  This more or less makes sure you have the latest version.
5. If you used git to get the code, run `git log -1` to see what commit you are on.  This is another way of checking that you have the latest code.
6. Run "ls" and make sure there are files named `lsm6.h` and `lsm6.cpp`.
7. Edit `prog_options.cpp` *in that directory* and change the default bus to `/dev/i2c-1`.
8. Run `make`.
9. Run `strings ./minimu9-ahrs | grep /dev/i2c` so we can see if it's using the right bus.
10. Run `./minimu9-ahrs --mode raw`.

Please tell me the results of each step.

closed 2021-07-21 by DavidEGrayson